atcollab/at #894: Wiggler Energy attribute error

T-Nicholls, 2025-01-21:
### Cause

If we create even the most basic Wiggler in Matlab it has the `Energy` attribute:
```
>> w = mywig("MPW15")

w = 

  struct with fields:

       FamName: "MPW15"
    PassMethod: 'GWigSymplecticPass'
        Length: 0
         Class: 'Wiggler'
            Lw: 0
          Bmax: 0
        Energy: 0
         Nstep: 5
         Nmeth: 4
            Bx: [6×0 double]
            By: [6×1 double]
        NHharm: 1
        NVharm: 0
```
So, if we then try to load a `.mat` lattice file containing a Wiggler in Python we get this error:
```
TypeError: at.lattice.elements.LongElement.__init__() got multiple values for keyword argument 'Energy'
```
This is because `Energy` was removed from the build attributes of the Wiggler element in #822 and so `Energy` doesn't get popped here anymore:
https://github.com/atcollab/at/blob/c8c58446715cf263fa94afc7a8990381e707138b/pyat/at/load/utils.py#L366
and so it isn't passed as the lowercase `energy` argument anymore. Instead, it's passed in as `Energy` with the other miscellaneous attributes in `kwargs`.
However, lowercase `energy` is also passed to `super().__init__()` as `Energy`:
https://github.com/atcollab/at/blob/c8c58446715cf263fa94afc7a8990381e707138b/pyat/at/lattice/elements.py#L1416-L1427
so we end up with an error because we are trying to pass two `Energy` keyword arguments to the constructor.

### Replication

Rather than having to boot-up Matlab and save a new lattice file, we can more easily replicate this directly from Python by doing:
```
>>> Wiggler("MPW15", 1.44, 0.06, 0.0, Energy=0.0) 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jzs47954/git/at/pyat/at/lattice/elements.py", line 1424, in __init__
    super().__init__(
TypeError: at.lattice.elements.LongElement.__init__() got multiple values for keyword argument 'Energy'
```

### Solution

The first question is why was `Energy` removed from the build attributes of the Wiggler element in #822 and a default value of `energy=0.0` added?
Assuming there was a good reason for doing that and so it shouldn't be reverted, then a simple solution would be to just pop `Energy` from the `kwargs` dict before that is passed to `super().__init__()`.
However, this raises another question - which energy value should be prioritised, the one passed as the `energy` argument, or the one passed as `Energy` in `kwargs`?
The question of which value should be prioritised when we have two arguments for the same thing was previously encountered in #695, with `K`/`H` and `PolynomB`. However, unfortunately, that issue got lost in my backlog and I never got around to discussing it with the wider community or fixing it, sorry @space-budgie. So I will create a discussion thread (#895) for us to decide how best to handle situations such as these.
Finally, there is a further question in this case, in my understanding, pass methods prioritise the energy value from the lattice anyway ignoring the `Energy` attribute of the element. so why do we bother with the energy argument & attribute for Wiggler & RFCavity elements if it's just ignored anyway?

lfarv, 2025-01-22:
Hello @T-Nicholls. Thanks for you analysis. I think that there are 2 different problems in your examples:
1. **Energy**. The beam energy is not an element property. It appears in Matlab because there is no "lattice" object to handle it. It is now preferably handled by a RingParam element, but RingParam is still not mandatory: old files still need to be interpreted. So originally the energy was repeated in each element needing it. The strategy for getting the energy in integrators is now clearly defined in #816:
   - in Matlab, the energy is taken from `RingParam` if existing, otherwise from the element. Since #816, `Energy` is optional in Matlab
   - in python, an element's energy is always ignored. So, as you mention, there is no reason to have the energy appearing in element constructors. However, nothing prevents users from adding custom attributes (`Energy` for instance) when constructing elements, by adding keyword arguments. These are ignored in PyAT, but may appear when reading an old Matlab file.
   
   Looking around, an "energy" argument still appears for compatibility reason in RFCavity and Wiggler. It should be ignored, and the Wiggler constructor is clearly wrong.

   When reading a .mat file in python, the strategy to get the Lattice energy is also clear: it is taken 1st from RingParam, (always existing for files created in python), then from the RF element if existing, and then from any other element.

2. **Double definition for the same quantity** ( `K` / `PolynomB[1]` for instance). Keyword arguments have priority over positional arguments. #695 was a bug and is corrected following this rule (I can't find in which PR)

lfarv, 2025-01-23:
I noticed that concerning energy, the wiggler is a very special case: it's the only magnet where the magnetic field is declared in "absolute", in Tesla. In all other elements, the field is normalised as $B/B\rho$. This is maybe a reason to keep a wiggler's 'reference" energy together with the peak magnetic field.

But at the moment, the Energy attribute of the wiggler, if present, is ignored in python, like in all other magnets. The Lattice energy is always used.

One possibility would be to scale the field when running a lattice with an energy different from the wiggler "reference" energy. That way, a wiggler would behave as the other elements. However this does not look so useful, since a wiggler is normally designed for a specific energy, it does not make much sense to scale it with energy.

As a first step, I propose to fix the wiggler bug in the following way:
- the energy will remain an optional attribute
- the energy positional argument stays for compatibility
- if given, a keyword `Energy` argument will take precedence over a positional argument called `energy`. That solves the problem of reading Matlab files
- This energy will still be stored as an element's attribute, even if it is not used in python

This does not change anything in the PyAT behaviour, it just solved the bug in a consistent way.
In the mean time, we can think of improving the behaviour: scaling, runtime warning if the lattice energy is different for the wiggler's one…

T-Nicholls, 2025-01-23:
Hello @lfarv,

Thank you for your explanations, they provide a much clearer view of the situation than what I had written.
However, one small point regarding #695, the bug of an error being raised  when creating a `Multipole` with both `poly_b` and `K` / `H` is still present on the master branch:
```
>>> Multipole("M", 0.5, [0, 0, 0], [0, 1, 2], K=1.5, H=2.5)
AttributeError: 'Multipole' object has no attribute 'PolynomB'
```
Should there be a difference between how `RFCavity` and `Wiggler` elements handle the energy argument/attribute? From my understanding of what you've said they should handle it in the same way. Currently, there are a couple of differences; so, I propose, beyond just fixing the bug with `Wiggler` we also make the handling consistent between the two classes.

lfarv, 2025-01-24:
Hello @T-Nicholls: thank you for summarising the present inconsistencies. While I agree with the goal of having a common behaviour, I can give some historical reasons for the present situation:

> - RFCavity has Energy in its _BUILD_ATTRIBUTES, Wiggler does not
> - RFCavity does not set a default value for the positional argument energy, Wiggler sets a default value of 0.0

Both are linked to the same cause: `Energy` is a mandatory argument in `RFCavity`, and it's optional in `Wiggler`. Why? Probably because in Matlab, before the introduction of RingParam, the energy which is necessary in the RFCavity integrator had to be there and was nowhere else. For wigglers, it was set as optional together with other radiating elements in #816. This ignored the fact that because of B in Tesla, the energy is required even without radiation… We can very well think of a common behaviour, but also decide how the `Wiggler` should behave when scaling the lattice energy.

> - the handling of energy and `RFCavity` & `Wiggler` elements is also different in multiple places inside `params_filter`

This is also related to the processing of old Matlab files: for a lattice without radiation, and before Wigglers were introduced, the RFCavity is the only element where `Energy` had to be defined. For a lattice with radiation, it was also present in all radiating elements, with possibly conflicting values. This is the reason for the precedence order in params_filter:

    energy is taken from:
        1) The params dictionary
        2) Cavity elements
        3) Any other element

The `Wiggler` is considered as any other radiating element. Here, to avoid any surprise in reading old Matlab files, I would prefer keeping this unchanged.

oscarxblanco, 2025-01-28:
Dear @T-Nicholls and @lfarv ,
`atinsertiondevicekickmap` use to have the same problem. At that time I chose to store the value in an argument called `Normalization_energy`.
```matlab
% matlab
Normalization_energy    energy to which the field table was scaled
```
This is backwards compatible with the element creation because the arguments are positional, but incompatible with saved lattices.

T-Nicholls, 2025-07-23:
Hello @swhite2401 ,
I have just proposed a fix for this on #960 - it is the simplest possible fix for this bug and doesn't change any other behaviour that @lfarv was concerned about here, and so hopefully conforms to what @lfarv had envisioned

commit 4c3ac10f (2025-08-15): Improve Wiggler energy handling to fix #894 (#960)

* No longer create Energy attribute for wigglers in lattice object energy setter
* Update Wiggler docstring to indicate energy is ignored
closed 2025-08-15 by lfarv