atcollab/at #915: Matlab atlinopt6.m with harmonic cavities

ZeusMarti, 2025-03-21:
Hi all,

We found out that in Matlab atlinopt6.m does not work well with lattices with harmonic cavities. In particular the lines below are a bit dangerous because it is assumed all cavities have the same frequency:

https://github.com/atcollab/at/blob/04e16f913162f9f9a8fd32e51f567f07f687a0eb/atmat/atphysics/LinearOptics/atlinopt6.m#L230-L235

Also when changing the frequency afterwards in the code it should change proportionally to the harmonic of each cavity.

Sorry if this is an already discussed issue we are planning to work on in the future.

Thanks for any info on this,

ZeusMarti, 2025-03-26:
@lfarv @swhite2401 ,

What about using: `f=min(freqs); ` instead of f`=freqs(1); `? I can't think of cases were there is any frequency below the main cavity. Assuming to always place the main cavity in the first RF location in the ring seems a bit arbitrary to me.

ZeusMarti, 2025-03-26:
For us this simple solution works: #916

lfarv, 2025-03-26:
@ZeusMarti 

> What about using: f=min(freqs);  instead of f=freqs(1); ? I can't think of cases were there is any frequency below the main cavity. Assuming to always place the main cavity in the first RF location in the ring seems a bit arbitrary to me.

You are perfectly right, and that's what is done in `atmaincavities`: it gives the index of all the cavities with the lowest frequency. Unfortunately, this function was not used in `atlinopt6`.

The solution proposed in #917 follows your comment above: the momentum change for computing the chromaticity is now done by scaling all frequencies by the same amount.

Side note: there is one case where the main frequency would not the lowest one: you can add an auxiliary cavity at h-1 to modulate the synchrotron frequency around the circumference (but h+1 does the same).

ZeusMarti, 2025-03-26:
Ah! Thanks @lfarv ! That is perfect, then I'll delete #916 .

lfarv, 2025-03-27:
Fixed in #917

closed 2025-03-27 by lfarv