atcollab/at #919: cavpts argument for lattice with changing number of elements

TeresiaOlsson, 2025-04-02:
The cavpts argument doesn't work correctly for a lattice when the number of elements are changed. For example, when splitting an element to put a marker in the middle. The user needs to remember to update the argument or it might not be possible to calculate the optics anymore.

I tried it and got the error ``AttributeError: 'Quadrupole' object has no attribute 'Frequency'`` because the cavpts were now pointing to a quadrupole instead of a cavity.

Is there a way where you can get this argument to always point to the cavity elements rather than to an index?

lfarv, 2025-04-02:
Hi @TeresiaOlsson. For which function is the `cavpts` argument you mention ?

TeresiaOlsson, 2025-04-02:
It's on the lattice.

I'm using the method that is suggested in the tutorial for how to specify which the main cavities are.

[https://atcollab.github.io/at/p/howto/CavityControl.html](https://atcollab.github.io/at/p/howto/CavityControl.html)

lfarv, 2025-04-02:
Ok, I see. Here, `cavpts`, like any "refpts-type" argument can be something else than an index.

It can be:
- an Element class: for instance `cavpts=at.RFCavity` will select all cavities, wherever they are. Not very useful here because it's the same as leaving out the `cavpts` attribute,
- a string, possibly including wildcards, specifying element names: for instance `cavpts="RF*"` will select all elements whose `FamName` matches "RF*". This will keep the right cavities even when changing the number of elements.

Otherwise, you can also think of not using the `cavpts` Lattice attribute. In such a case, the `set_*_cavity` functions act on all cavities, **assuming that the main cavities are the lowest frequency ones**. The values you specify are sent to the "main" cavities (lowest frequency) and the other ones are set so:
- the frequencies are scaled by the same amount as the main cavities,
- the voltages are scaled by the same amount as the main cavities,
- the timelags are shifted by the same amount as the main cavities.

In most cases, that's what is desired.

TeresiaOlsson, 2025-04-03:
Ah! Thank you. I didn't realise that I could define it as a string. But it makes perfect sense that you can use it the same as everywhere else. Now it's works perfectly also when the number of elements are changing :)

TeresiaOlsson, 2025-07-14:
@lfarv I changed in my pyAT lattice to use a string to define `cavpts`. Works great in pyAT! Problem now is that if I save as mat file and try to use the lattice in matlab it doesn't work. I run into problems in `atsummary` on line 55

```
 case 'rf_frequency'
    v=get_unique(ring(getmaincav(ring)), 'Frequency');
```

I guess because the matlab version requires indices and can't handle finding elements by string?

TeresiaOlsson, 2025-07-14:
No. Seems like I was wrong. Sorry, that part seems to still work. Something else is broken...

closed 2025-07-14 by TeresiaOlsson

lfarv, 2025-07-20:
Hi @TeresiaOlsson 
You are right, moving Lattice indices between python and Matlab is tricky, for all the reasons you mentioned:

- Matlab can only use integer or boolean indices (I don't think that single- or double-quoted strings makes any difference),
- Matlab has usually an additional RingParam element,
- Even worse: Matlab numbers array elements from 1 and python for 0.

I do not see a general solution for that. One could make a special treatment for the `cavpts` Lattice attribute: when saving a .mat file in python, convert to boolean indexing and take RingParam into account. But that would loose the "string" information if using the .mat file for python to python transfer.

Or, one could think of saving in .mat files a python-specific block of information ignored by Matlab. This means a large modification of the .mat format.

The problem is the same for other file formats like .m or .json.