atcollab/at #935: Lattice parameters when no cavity in lattice

TeresiaOlsson, 2025-05-16:
Hi,

In the pyAT documentation for radiation_parameters it says "Valid for uncoupled lattices with no RF cavity or radiating element." but if I try to use it for a lattice without a cavity inside I get the error "AtError: No cavity found in the lattice" because of the line "voltage = ring.rf_voltage".

Why is there a requirement to have a cavity in the lattice if the function should be valid without RF cavity? And if radiation_parameters can't be used, what is the correct way to calculate the lattice parameters for a lattice without a cavity? For example, during the early stages of lattice design.

lfarv, 2025-05-19:
@TeresiaOlsson :I see the problem, I'll look at that…

lfarv, 2025-06-24:
fixed in #943