atcollab/at #987: madx/mad8 parser ignore vertical dipole (normal dipole with TILT)

simoneliuzzo, 2025-09-04:
A small issue concerning lattice parsing.

A vertical dipole defined like this in mad8/X

VSEP1: RBEND, L=4.0, ANGLE=KVSEP1, TILT=-1.570796326795

is translated in at ignoring the tilt, that is rather relevant as it transforms the dipole in a vertical dipole.

I tried to use set_tilt and set_rotation but the change does not appear as expected.

thank you for your help in sorting this 

best regards 
Simone

kparasch, 2025-09-05:
Hi Simone, if I am not mistaken, your AT solution would leave the reference orbit unchanged while MADX would redefine it according to your angle. Or is this not the case?

This seems related to Issue #781

lfarv, 2025-09-07:
Hello Simone,

I see the problem: in PyAT, tilting an element is a rotation around an axis joining the entrance and the exit point. For straight magnets, it's an S rotation at entrance and the opposite rotation at exit. But for a curved magnet, it's more complicated and it involves rotations around the 3 axes at both ends. And as @kparasch pointed out, it does not change the reference trajectory.

While in MAD-X, it is just an S rotation at each end.

So the problem is easy to fix, I'll open a PR soon.

lfarv, 2025-09-07:
@simoneliuzzo: concerning your other remark about the unexpected difference between `PolynomA[0]` and `KickAngle[1]`, I'll look at that.

But in any case, this is different from a vertical bend: your `Multipole` is a straight magnet which introduces a large orbit deviation but keeps the reference orbit straight, while a rotated dipole will bend the reference orbit but introduce no orbit deviation.

simoneliuzzo, 2025-09-09:
Dear Laurent,

thank you for your reply.

In fact we checked and the vertical dispersion generated by the kick and polynomA/B is the same.  the issue is only that a rotated dipole does not rotate the reference trajectory accordingly. 

I will test your modifications just now and let you know.

Do you think also geometry and plot_geometry should be modified following this change?  (to take in account dipoles with tilts and produce "3D surveys")

thank you!

best regards
Simone

lfarv, 2025-09-09:
> Do you think also geometry and plot_geometry should be modified following this change

Matlab has the at geometry3 which does that. For sure it would be interesting. But that's another topic…

For now, I close the issue.

closed 2025-09-09 by lfarv